AbkarinoMHM/PS4SysconTools #3: Add support for Arduino nano v3

galacticwarrior448, 2023-06-28:
Hello, would you consider adding support of being able to do this with an Arduino nano v3 with ch340?

las-sign, 2023-06-28:
That would be great, Arduino nano v3 with ch340 is more common and affordable then teensy boards. 

Darknesmonk used Arduino nano v3 (even 168p) and just 150 lines of code for reading native syscon. 
Also he has paid version for writing it (like BwE's). 

So I bet it's quite possible technicaly. Maybe not so fast as teensy, but who cares, when it's the cheapest way.

Anyway thanks for your effords, free tools is moving the scene

AbkarinoMHM, 2023-06-30:
Hi all,

our initial project was passed on Arduino NANO v3.0, but during testing we found that it is not stable enough unless you use more external components like optocoupler and some other resistors, also our plan was to add a lot of new features that Arduino can not support right now.
But maybe if we have a free time we can return to old design and improve it.

tekaikko, 2023-07-04:
Hi!
Just about any level of support for Arduino would be helpful. I am currently using Renesas Flash Programmer to program original RL78 chips, using the basic serial programming circuit provided by Renesas. If you could just share a minimum Arduino version to unlock reflashing of the "sony" branded RL78, it would be great. I mean, if it is possible, to unlock the chip using your "trick", but keep using Renesas software to do the actual programming...

andy-man, 2023-07-06:
Great project [AbkarinoMHM](https://github.com/AbkarinoMHM)!
It would be realy cool if you make it possible for Arduino nano v3, some extra components is much cheaper then teensy boards :)
Recently I've started project called [PS4 Wee Tools](https://github.com/andy-man/ps4-wee-tools) for NOR and Syscon dumps manipulation which could be free alternative for some of BwE apps
So with help of both our projects the downgrade would be much affordable

AbkarinoMHM, 2023-08-02:
Hey @andy-man 
I'm so sorry for the late reply, i just saw your comment now, we are working in porting our code to more modern and cheaper dev board based on RP2040 (RPi Pico), it will be more stable and powerfull than Arduino Nano v3 and will not require a lot of external components.
Btw thanks so much for your nice project [PS4 Wee Tools](https://github.com/andy-man/ps4-wee-tools) it is amazing, and we can co-operate also if you want.

tekaikko, 2023-08-02:
Thanks, perfect, RPI pico is readily available and cheap!!

AbkarinoMHM, 2023-08-04:
Thanks all for your suggestions and comments, now we will consider this issue as closed, until we port our code to Arduino Nano v3.

closed 2023-08-04 by AbkarinoMHM

magwangkh, 2023-08-22:
Hallo ABbkarino.. first i start by saying thank you and Allah bless you for all the the effort you put in this software for it will help alot of people... May HIS mercy be upon you and May HE bless the work of your hands.. am sorry to ask whats my best option? i buy the Teensy 2++ or i wait for Arduino nano v3 port?

Lereda, 2023-11-10:
Hi AbkarinoMHM i wanted to ask a question but before i ask the question i wanted to say that you are really talented to make such a good tool for the ps4 and that i wanted to say thank you for that, and my question was if it is possible to make the code (syscon reader) able to work on rp2040 pico and if yes how i could do it ? Thank you in advance.

dazzaXx, 2025-05-13:
Is there any update on this?