AbkarinoMHM/PS4SysconTools #11: Program freeze on connect

kvdCode, 2023-07-28:
Hello everyone, this is my first post on github. I try to resolve my problem alone with a lot of search on internet but I failed...
Here my problem:
Programmed teensy 2.0++ with good hex file but when I connect to my teensy serial port, program freeze on startup and I never have the succefully connected popup message. When I disconnect the D2 pin from teensy board, I have the connection error popup and program is responsive.

Maybe stuck in the clsPS4SysconTool.cs while loop at line 129
while (serialPort.BytesToRead < 4)
{
   System.Threading.Thread.Sleep(10);
} 

I rechecked three times all my traces from syscon pins to teensy mcu pins for continuity and shorts with my binocular stereo microscope. Tested with 100-150-250 ohms resistors with and without 3.3v teensy regulator because of RL78 5v tolerant from Renesas datasheet. Tested on two different computers also...

When I try to connect to teensy serial port with a 115200 baudrate, console return #E0 on any typing key. Maybe teensy waiting for a command to start glitching ? or maybe a closed door from syscon ??

Do I have this freeze because of my OCD rom not version 3.03 or my chip not supported by this software ?

Anyone help will be greatly appreciated.

Thanks

model: PS4 PRO 2018 - CUH-7216B
current fw: 9.03
prev fw: 9.00
syscon chip: 64 pins - A06-COL2
glitching method: onboard + teensy 2.0++ (new one just received from aliexpress)

kvdCode, 2023-07-30:
I reply to myself for others who will read this post later. So I succeeded to connect syscon to my computer but I had to
completely desolder the syscon chip. After that, 3.3v not worked at all for me so I turn teensy to 5v and everything works well. But no luck for me, my PS4 PRO whith 9.03 firmware and 8.50 before had already downloaded 9.60 and turned the slot
so game over for me. BUT... now I resolved syscon 
 issue, I have a BLOD on power on. I don't know why. I activated uart in the nand but
I got absoutely no ouptut on TX pin. Another problem after the first. If anyone have an idea? Thanks

AbkarinoMHM, 2023-08-02:
Hey, it seems a connection issue or your setup was wrong at the begining, and regardsing BLOD, can you tell me if you tried to dump/flash your SPI/NOR flash or not? if you did, please check all resistors around SPI flash first with multimeter.

woooow1, 2023-08-11:
I also have this issue! With current 2.0.0 version. I decided to swith back to 1.0.0 and connection is ok. Use teensy 4.0. Syscon is soldered to board and pin 22 is lifted...

woooow1, 2023-08-11:
I did more investigation for problem. I think, is firmware v2.0.0. With firmware 1.0.0 i made connection with both softwares. v1.0.1 read and write chip, 1.5.0 can only read. With firmware 2.0.0, no connection on both software versions!

AbkarinoMHM, 2023-08-13:
Thanks @woooow1 for reporting this issue, i'll check teesny 4.0 firmware, since i did not have teensy 4.0 my self, @egycnq should maintain this code, but he is not availabe right now, so i'll review it and update it with new fix.

tttheslimshadyyy, 2023-08-28:
Same issue on teensy 4.1 
I programed teensy 4.0 hex file to teensy 4.1 
It reads good but gives error while writing and it breaks syscon chip.

AbkarinoMHM, 2023-09-14:
Hi All,
Please updated to the latest version "FW v2.1" then try again, i had fixed some issued on Teensy 4 and 4.1 firmwares.

woooow1, 2023-11-06:
Thanks @AbkarinoMHM, verson 1,60 with hex files v2.01 works perfect! Keep that good work!

Mladicbt, 2023-12-29:
Did you figure out the blod I have same problem

AbkarinoMHM, 2024-01-28:
Thanks @woooow1 for your feedback, now we consider this issue as closed.

closed 2024-01-28 by AbkarinoMHM