AbkarinoMHM/PS4SysconTools #12: restore to R5F101LLAFB

shoyrock, 2023-07-31:
hey just wanted to know if there's a way to write syscon back to an R5F101LLAFB, I've been trying without any lock just wanted to know if I'm doing something wrong

AbkarinoMHM, 2023-08-02:
Hey, if you mean a stock chip, then our tool will not help you, since the stock chips i snot protected by the key that we are using to access original PS4 Syscon.
You can find any cheap renesas flasher or build your own with just 2 diodes and USB to TTL converter.

shoyrock, 2023-08-02:
got it thanks for the help

shoyrock, 2023-08-03:
> Hey, if you mean a stock chip, then our tool will not help you, since the stock chips i snot protected by the key that we are using to access original PS4 Syscon. You can find any cheap renesas flasher or build your own with just 2 diodes and USB to TTL converter.

hey I tried looking up how to go about doing this but I wasn't able to find anything about it online, would you be able to post a link to 2 diodes and USB to TTL converter?

AbkarinoMHM, 2023-08-04:
Yea, you can refer to this project, it is open source hardware/software for renesas mcu
https://github.com/msalau/rl78flash

AbkarinoMHM, 2023-08-04:
You can refer to this simple diagram for hw flasher:

[image]

for me I was using mode 3 and it was working fine.

closed 2023-08-13 by AbkarinoMHM