AbkarinoMHM/PS4SysconTools #18: Erased flash

marcinmaster999, 2023-10-12:
Hi. When I erased all the memory using your tool, communication is correct until ocd id, then the chip responds 0xF3 - Checksum mismatch. When I change ocd id to 0xff it responds 0xF4 - Checksum (matched but OCD ID not ?) I have no idea how to fix it ?

AbkarinoMHM, 2023-10-16:
Hi, are you talking about original PS4 Syscon chip? if you completely erased it, then my tool will not be able to handle it again unless you wrote firmware back before restaring the chip or at least you have to write the boot0 blocks.

AbkarinoMHM, 2023-10-16:
May i know why you completly erased your chip?

marcinmaster999, 2023-10-17:
Yes, this is the original syscon. It was probably deleted by mistake, as I reported earlier, the program crashed, in this version this problem does not occur. I know that I cannot recover it using your tool, so I connected it to Arduino and it stops at 0xF4, i.e. the wrong key.

woooow1, 2023-11-08:
In version 1.6 option "erase before write" not work!

AbkarinoMHM, 2025-05-12:
All this issues should be covered in the last release so please use it.

closed 2025-05-12 by AbkarinoMHM