AbkarinoMHM/PS4SysconTools #19: v1.5 and 1.6 not working on teensy4

boxochi, 2023-10-19:
hi i have teensy4 and v 1.00 hex working good but version 1.5 and 1.6 not working
when i open the tool its freeze on connecting to teensy !
i checked again with 1.00 version and its connected as well

ebedevelopment, 2023-10-19:
strange, a friend of mine had tested v2.01 on teesny 4.0 also and it work without any issue, i did not have tessny 4.0 on my hand to test but i'll check it with my friend, anyway thanks for your report.

RobbedColek, 2023-11-07:
2.01 with SYSCON Tools 1.6 works fine with my Teensy 4.0.

Setiasetiawan, 2023-11-20:
hi, I'm using teensy 4.0 with hex 2.01 but it freezes when first connected (on ps4 pro cuh70), while on ps4 slim it runs smoothly

ng1n33r, 2023-12-13:
PS4SysconFlasher_Teensy_4.0_v2.01.hex works with Teensy 4.0 on fat PS4 (syscon FW 1.00)

closed 2024-01-28 by AbkarinoMHM