AbkarinoMHM/PS4SysconTools #22: Feature Request: Ability to Write to New/Blank Syscon Chips

ng1n33r, 2023-12-14:
Great tool!

It would be great if the devs could add the ability for the _PS4 Syscon Tool_ to access/write to new/blank syscon chips.

This would allow users to:
- Use new/blank RL78 chips
- Write to their original syscon chips, in the event of a disconnect/freeze/power loss/etc... after/during a full erase

I'm aware of, and I have used [rl78flash](https://github.com/msalau/rl78flash). It would be appreciable if the _PS4 Syscon Tool_ can do it as well.

Cheers

bosaad21, 2025-04-14:
Any plan to add this feature soon? Otherwise How I can use rl78flash? I got new blank syscon so I need to write my syscon bin backup..

AbkarinoMHM, 2025-05-16:
Current you can use the [rl78flash](https://github.com/msalau/rl78flash) tool without any issues, i had used it my self many times, since our tool was mainly designed to flash a write protected PS4 Syscon chips, but we can consider this in the upcoming releases.

Regards,
Abkarino

closed 2025-05-16 by AbkarinoMHM