AbkarinoMHM/PS4SysconTools #23: Is debug mode deactivated after an update?

ArturSchnell, 2023-12-24:
Hello everyone,
I have previously been able to use the tool to read out my Syscon and have also activated the debug mode to be able to connect without cables. I thought that because I now have a backup, I can revert to this version at any time. So I did an update to sync my trophies to PSN. 
Then I wanted to go back to the firmware from my backup. There were no problems with NOR, but I can no longer connect to the Syscon... 

So my question: Is it possible that the update rewrites the Syscon in such a way that the debug mode is reset?

ebedevelopment, 2024-01-28:
Hello,

I did not think so, because the Syscon debug flag is not updated by PS4 console during firmware updates, so maybe something wrong with your setup.

Best regards,
Abkarino

AbkarinoMHM, 2024-01-28:
Hello,

I did not think so, because the Syscon debug flag is not updated by PS4 console during firmware updates, so maybe something wrong with your setup.

Best regards,
Abkarino