AbkarinoMHM/PS4SysconTools #24: r5f100fcafp (Renesas RL78/G13) 44pin

sedeco99, 2024-01-27:
hi i need read this r5f100fcafp (Renesas RL78/G13) 44pin  by your software  but no connect  can any tell me thanks

ebedevelopment, 2024-01-28:
Hi,
you should use any RL78 flasher for that, our tools only support original PS4 Syscon (RL78/G13) chips not a stock chip like in your case.
you can refer to this project that will help you: https://github.com/msalau/rl78flash

Best regards,
Abkarino

AbkarinoMHM, 2024-01-28:
Hi,
you should use any RL78 flasher for that, our tools only support original PS4 Syscon (RL78/G13) chips not a stock chip like in your case.
you can refer to this project that will help you: https://github.com/msalau/rl78flash

Best regards,
Abkarino

sedeco99, 2024-01-29:
hello thanks my dear  i thing rl78/g13 all same  to read   because all programmer can flash and write to chip but no programmer 
can read this   thanks to you AbkarinoMHM  me  egypt  ahmed

MohamedHassanNasr, 2024-03-17:
you are always welcome Ahmed, thanks so much for your too.

MohamedHassanNasr, 2024-03-17:
by the way reading feature is limited to specific RL78/G13 boot ROM version 3.03 and also with specific unlock code :NotUsed:, that is why i said it will not support blank chip unless it have the same requirements.

sedeco99, 2024-03-17:
> by the way reading feature is limited to specific RL78/G13 boot ROM version 3.03 and also with specific unlock code :NotUsed:, that is why i said it will not support blank chip unless it have the same requirements.

thanks my dear some programmer like jlink flasher  read  but very high price  and some programmer in car dash do this

riegaz, 2024-04-04:
How likely it is that another car dash RL78 uses OCD 3.03? What do you think? Is there any other tool around that would let me read a R5F100? It would be great if you could point me in the right direction.

Anything else I could do to read the flash of a R5F100?

sedeco99, 2024-04-06:
hello   lock here  and install software you can see the programmer can read flash and code tor this ics 
https://www.segger.com/products/debug-probes/j-trace/models/j-trace-pro/

closed 2024-04-23 by AbkarinoMHM