AbkarinoMHM/PS4SysconTools #28: Anyway to recover?

TheCroupier74, 2024-04-16:
Hi I was using V1.60 and successfully backed up my syscon.

I then tried to write back the backup but the tool immediately froze, non responsive. I waited a while then had to force quit.

I tried to reconnect to read the chip but it fails.
looks like the chip erased with no data written.

I understand your tool cannot write to a blank chip?
What are my options to recover from this issue and in the future is there anyway to prevent these random freezing of your software?

Thanks.

alpay35, 2024-04-16:
Have you tried programming externally by inserting the sysconn chip?

TheCroupier74, 2024-04-16:
No, on board with pin 22 lifted.
I think my cheapest option is to buy a used syscon, BWE’s Syscon rebuilder could probably fix it but it’s $175, I could just buy another PS4 for that cost.
Luckily I have a good backup of my Syscon.

AbkarinoMHM, 2024-04-22:
Sorry to hear that, but in worest case, if you have a good backup, you can write it back to any syscon chip eather blank or originally locked one, and it will work without any issue.

TheCroupier74, 2024-04-22:
Yes I have backup, I dump 2 times, verify they are good then made extra copies. Will look for a cheap replacement Syscon chip.

AbkarinoMHM, 2024-04-23:
Great to hear that, you can get a any chip from a dead board it will be very cheap and easy to find.

closed 2024-12-05 by AbkarinoMHM