AbkarinoMHM/PS4SysconTools #29: syscon image rebuild

canariaslink, 2024-04-25:
i've got a ps4 very cheap with a a05-col2 chip , i installed the teensy2++ and erase by mistake , what data and on wich offset are per console keys , because i writed another syscon image from other console and no beep no nothing..... , i want to attempt to recover it i installed a new syscon and i'm able to read and write now

AbkarinoMHM, 2024-04-25:
Did you made a full backup first or not?

canariaslink, 2024-04-27:
no , the flasher crash and corrupt the data inside , so i only have the first block , after that , locked , i swapped it to another one and make a new image with svns factory reset and 2.29_4 image and the console attempt to boot , ofc without uart support , maybe fw issue , i'm trying so many fw , u know what fw have from factory a a05-col2? thanks so much for your tool , makes the life easyer your method

canariaslink, 2024-04-27:
i wanna attempt to trigger factory mode wich bypass the check of svns areas

AbkarinoMHM, 2024-04-27:
> no , the flasher crash and corrupt the data inside , so i only have the first block , after that , locked , i swapped it to another one and make a new image with svns factory reset and 2.29_4 image [...]

Did you tried to dump it first, even if flasher crashed during dumping process it will not kill the chip, i think that you tried to flash it directly without having a working backup first.

AbkarinoMHM, 2024-04-27:
> i wanna attempt to trigger factory mode wich bypass the check of svns areas

Good luck, this is not possible without real PS4 FSM USB Dongle/Flash, if you have an official Sony repair center in your country, you can try to send your console to them for repair.

canariaslink, 2024-04-28:
this is wrong , in this state the console don't read usb , don't power just a little bit blue light with some patches i made and shutdown , so how will read a usb if the system not pull up? i'm miralatijera aka dospiedras1973 i'm not a n00b....

canariaslink, 2024-05-02:
> hi my problems is: secure loader build: Jun 8 2021 05:20:42 (r10426:release_branches/release_08.520) [711MHz] AGESA: ThebePBDK W5C21 standby 05550000 ERROR: main(3738) checkUpdVersion 0x5560000 != 0x8520000 please help me

u are attempting bad slot , this is why , i recomend on syscon use the patch C or B and on NOR flash 
btw everytime u boot up the console some syscon's flags are set's , so it's better all time don't mix the dumps for example , if you bootup and only dumped the syscon maybe u have problems later with the bank's settings , ps4 security are constantly aplying patches to syscon , and on nor flash are checked , this error u have means one of the slots are bad flashed , you need to match the snvs slot wich match with your older version , seems the nor flash one are ok because ( release_08.520 ) try all patches generated by the ps4-wee-tools ( the free ones , it's a abuse 120usd for a software wich u will use 5 min ) and generate your correct syscon image , ( every try reflash the nor flash with  the base one for make sure in the boot any flag are configured on the nor flash because a bad boot register )

Alexon11, 2024-05-02:
здраствуйте. Удалось решить вопрос с сбивающимся временем ? я заметил что время обычно слетает: при снятии микросхемы Bios c платы. Второе через debag menu не работает исправление, верменно решается через ручную настройку до тех пор пока приставку не отключить от сети. Исправление у меня не сработало при попытке сделть копии bios и syscon не снимая чипы с памяти, обновления до новой версии , синхронизации и возврата бекапа не снимая микрухи. Тут у мня только предположение что нужно приствуку по новой понижать через переключение слотов и патча syscon.

Intruder911, 2024-09-16:
> > i wanna attempt to trigger factory mode wich bypass the check of svns areas
> 
> Good luck, this is not possible without real PS4 FSM USB Dongle/Flash, if you have an official Sony repair center in your country, you can try to send your console to them for repair.

@Hi, I have PS4 1000 model in FSM. Is there is any way to out from FSM? It was on very old 3.50FW, but with error su-39176-6 error on installation. So i reverted it by your tools to previous 2.02FW and had success FW re-installation. But as i said it is on factory / service mode. I think it was gone on factory/service mode in authorised service center Sony. Because there is no sign of soldering and UART was activated. No signs of soldering on UART pins and other pins. The PS4 and motherboard is as just from factory.

ogehen, 2025-02-24:
error su-39176-6 # I was able to solve it. By activating EMBOOT BD ROOM by injecting new chips, changing a little in the SVNS syscon area, the console can continue the firmware installation process and it is successful

Intruder911, 2025-02-26:
> error su-39176-6 # I was able to solve it. By activating EMBOOT BD ROOM by injecting new chips, changing a little in the SVNS syscon area, the console can continue the firmware installation process and it is successful

Hi. Can you explain how exactly please? THX

AbkarinoMHM, 2025-03-04:
> > > i wanna attempt to trigger factory mode wich bypass the check of svns areas
> > 
> > 
> > Good luck, this is not possible without real PS4 FSM USB Dongle/Flash, if you have an official Sony repair center in your country, you can try to send your console to them for repair.
> 
> [@AbkarinoMHM](https://github.com/AbkarinoMHM) Hi, I have PS4 1000 model in FSM. Is there is any way to out from FSM? It was on very old 3.50FW, but with error su-39176-6 error on installation. So i reverted it by your tools to previous 2.02FW and had success FW re-installation. But as i said it is on factory / service mode. I think it was gone on factory/service mode in authorised service center Sony. Because there is no sign of soldering and UART was activated. No signs of soldering on UART pins and other pins. The PS4 and motherboard is as just from factory.
> 
> Thank you

Hi,

  If you still did not solved your issue and still have the original sflash/syscon dumpes before downgrade please share it and i'll do my best to solve your problem.

Abkarino

Intruder911, 2025-03-08:
> f you still did not solved your issue and still have the original sflash/syscon dumpes before downgrade please share it and i'll do my best to solve your problem.
> 
> Abkarino

Hi. No, not solved. Here the files

AbkarinoMHM, 2025-05-17:
> > f you still did not solved your issue and still have the original sflash/syscon dumpes before downgrade please share it and i'll do my best to solve your problem.
> > Abkarino
> 
> Hi. No, not solved. Here the files
> 
> https://drive.google.com/drive/folders/1chV0Xsp5VOcR9jlEwemRRP_i5v6AE9_s?usp=sharing

So this is the original dump on FSM without any patches right?
I'll try to see if we can do anything with it or not.

Intruder911, 2025-05-17:
> i wanna attempt to trigger factory mode wich bypass the check of svns areas

Great. I'll try provide dumps and etc if you'll need for this.

closed 2025-08-06 by AbkarinoMHM