AbkarinoMHM/PS4SysconTools #31: Cannot enter debug mode anymore

mrkylegp, 2024-05-03:
Everything was fine, I took two dumps.

I then used the Enable debug mode option and wrote to the syscon. I got an error.

Now no matter what I do, the syscon will not go into debug mode again.

Is it bricked?

josearr, 2024-05-06:
Same exact issue here.

The error recieved was something about "maxium value, minimum value, expected 3"

mrkylegp, 2024-05-06:
> Same exact issue here.
> 
> The error recieved was something about "maxium value, minimum value, expected 3".
> 
> The error I receive now when trying to do a dump or whatever is "Error while exploiting PS4 syscon".
> 
> If using PS4 Wee tools, it shows there "unknown status code [0xE0]"

Yep that sounds like the error I got. The syscon IC is bricked. I will not be using this tool again. However I'll leave this issue here for the devs to improve on.

josearr, 2024-05-06:
@mrkylegp Did you get any help anywhere else to know for sure that the syscon is bricked?

What's the expected error if you try to turn on the PS4? I'm not getting power at all (no beeps, no lights, no fan movement) but I highly doubt it has to do with that, I was expecting some kind of error on screen.

mrkylegp, 2024-05-06:
> @mrkylegp Did you get any help anywhere else to know for sure that the syscon is bricked?
>
> What's the expected error if you try to turn on the PS4? I'm not getting power at all (no beeps, no lights, [...]

Numerous reports in Reddit of this also happening to them. Bwe also tweeted that this software bricked syscons in his testing. Regardless of your opinion of bwe, this seems concerning.

You won't get any response from the system now because the boot0 information on the syscon has been corrupted so the syscon is effectively brain dead now.

efargas, 2024-05-17:
Same problem here but i have the backups, it's possible to flash them on blank syscon or enter debug mode with bwe uart glitch?

rutdarkronin, 2024-06-03:
Sh*t, wish I've saw this thread before...
Another dead syscon here, though I got that dump safe and sound. Another swap job...

rutdarkronin, 2024-06-03:
@AbkarinoMHM, please make appropriate warning statement at release page because issue is common and folks will literally corrupting their syscons wildly! (Just like me)

josearr, 2024-06-04:
@efargas I don't think BWE solution will work once it's bricked. I contacted them in their discord and they only gave me the replace syscon from donor board solution.

I tried flashing my bricked one with my dump using rl78flash tool but none of the write operations were "acked" by the chip, I guess it's write protected without the glitch. I could read the content of the chip though and it detected that there was a discrepancy in the boot0 blocks that this shitty tool corrupted.

I have gotten a syscon from a donor board but I'm still thinking about what to do because I don't want to spend more money buying a hot air station for thsi and I don't feel like soldering 64 wires to the board and the syscon with the iron solder.

CodeKill3r, 2024-06-16:
So why is there no method to recover from a brick?
I also bricked one: seems it erased block0 (0xff-ing the config bytes, and the password) so this tool cannot enter OCD mode anymore -- just changing the password does not worked, tried hexediting the teensy fw. (Maybe different payload or need to find another way in? -- how the ROM code handles these config values?)
rl78flash also refused to do anything, as it cannot (denied to?) erase the rest of the flash and seems only able to write after.

As a side-note: as the NVS/SNVS are in later blocks, and the fw-dumps are available, if entering OCD would be possible, it could be possible to recover even without a full dump (just find the matching fw, and rewrite block0).

rutdarkronin, 2024-06-18:
> So why is there no method to recover from a brick? I also bricked one: seems it erased block0 (0xff-ing the config bytes, and the password) so this tool cannot enter OCD mode anymore -- just [...]

There is no known way of flashing stock syscon chips externally apart from downvolting glitch attack which this and bwe tools uses both. That's why it's a dumpster once you erase boot0 and powercycle. Syscon won't start itself after that cause of missing bootloader, won't initialize internal memory and you cannot access it anymore.

CodeKill3r, 2024-06-18:
> There is no known way of flashing stock syscon chips externally apart from downvolting glitch attack which this and bwe tools uses both. That's why it's a dumpster once you erase boot0 and [...]

Not true: it enters programming mode as evident by the rl78flash communicates with it. It queries blocks (blank check), get replies, but does not let other blocks be erased (implied erase-protection?)
 It just in a state where it can't be read/erased/written (maybe need some hidden command not implemented in rl78flash?)
The thing is, by the datasheet if FF-ed the 000C3H it set the OCDenabled and NoEraseOnAuthFail bits --- buut missing the "magic bits" ( 000010b) between and it's not explained what it means. Will it default to locked? Check the alternate address? I try to understand what the ROM code does (dumped form another board), but not found the relevant parts yet.

 Also please don't forget, it all started from fail0verflow's blog and wildcard's sysglitch PoC where the resources are.

CodeKill3r, 2024-06-22:
So there is a check in OCD_CONNECT (0x91) after the password (and chksum) sent and reply with 0xF4 then halts the CPU if the pw was blank (all 0xFF)
Glitching after sending the chksum may help avoiding this trap.

```
ROM:EFEB2                 CALL    tool_rx_0              ; read chksum byte
ROM:EFEB5                 INC     A
ROM:EFEB6                 SUB     A, C
ROM:EFEB8                 BNZ     tx_pwrefuse         ; chksum invalid
ROM:EFEBA                 CMP0    X                         ; check pw match
ROM:EFEBB                 BZ      pw_match
...
ROM:EFED7 pw_match:
ROM:EFED7                 MOVW    HL, #0C3h
ROM:EFEDA                 MOV     B, #0Ah
ROM:EFEDC                 MOV     A, #0FFh
ROM:EFEDE
ROM:EFEDE loc_EFEDE
ROM:EFEDE                 AND     A, ES:[HL+B]
ROM:EFEE1                 DEC     B
ROM:EFEE2                 BNZ     loc_EFEDE
ROM:EFEE4                 INC     A
ROM:EFEE5                 BNZ     pw_not_blank
ROM:EFEE7
ROM:EFEE7 pwbad_erase_not:                          ; this is where the pw mismatch with OCDERSD=0 enters too to 'wipe flash'
ROM:EFEE7                 MOV     A, #0F4h
ROM:EFEE9                 CALL    tool_tx_0_
ROM:EFEEC
ROM:EFEEC loc_EFEEC:
ROM:EFEEC                 HALT
ROM:EFEEE                 BR      loc_EFEEC
```

or if there is somewhere an erase_chip command that disregards block protections.... (rl78flah uses sequential block erase even in that case)