AbkarinoMHM/PS4SysconTools #37: Teensy Alternatives, like ESP32 - Raspberry Pico?

wtfrules, 2024-09-08:
Hi There!

Is it possible to modify the code to support ESP32 and/or Raspberry Pico for read/write the SYSCON?

Thanks!

BetaH4x, 2024-09-16:
Hi @AbkarinoMHM, @egycnq,

Also, as Teensy 2.0++ and Teensy 4.0 are compatible with Arduino libraries, compatibility with any ATmega328P MCU (Arduino Nano, Uno, etc.) will be easy and great for increasing the number of boards that can be used with it. Also, I would like to have compatibility with Arduino Nano Every, based on ATMega4809 MCU.

If you don't currently have the time and resources to do it, I can offer myself to work on porting the code to those platforms.

Thanks!

gameoflord012, 2024-10-05:
Hi @BetaH4x 

I don't have access to Teensy 2.0++ and 4.0, base on your last reply, does **Arduino Nano V3.0 ATmega328P** a good candidate?

Also if Arduino **Nano V3.0 ATmega328P** is compatible, how differently to wire it compare to Teensy 4.0? Is there any diagram for it or at least a topic I can start to research on.

Much thanks.

egycnq, 2024-10-05:
> Hi There!
> 
> Is it possible to modify the code to support ESP32 and/or Raspberry Pico for read/write the SYSCON?
> 
> Thanks!

Hi, I'm porting the code to raspberry pi pico and I can tell yes we can support it, still didn't try ESP32 but let's have a look soon

egycnq, 2024-10-05:
Hi @gameoflord012 @wtfrules @BetaH4x , I'm porting the code to raspberry pi pico and I can tell yes we can support it, still didn't try ESP32 but let's have a look soon. For Arduino Nano we found it unstable because it had only one HW UART interface and to fix it we will need to use more components with the Nano to increase the stability so for the moment I think it's better to port it to Raspberry Pi Pico a faster and more stable choice. I want to give a special thanks to @BetaH4x for your interest in our project

gameoflord012, 2024-10-05:
Hi @egycnq, thank you for replying. I have a few more questions.

Does the current release work with raspberry pico? If so is there any diagram how can I wire it with ps4 fat? Or at least a topic I can start to research on.

Much thank!

egycnq, 2024-10-05:
> Hi @egycnq, thank you for replying. I have a few more questions.
> 
> Does the current release work with raspberry pico? If so is there any diagram show how can I wire it with the ps4 fat? Or at least a topic I can start to research on.
> 
> Much thank!

The current release not support Pico yet, once it done and well tested it will be released 😊

jmw6773, 2024-10-07:
@egycnq,

  I don't know how far you are in the porting process, but I'm excited to hear you are working on it. Porting it to the Pico or ESP32 chips lowers the cost for users that just want a one-off tool. I have a PS4 in my closet with a 'loadBios -8' error that passes RAM checks and have been debating purchasing a Teensy to try to fix it. I don't have any other use for the Teensy, so the price seems steep. But I already have a Pico and several ESP32's around the house I could use.

  If you get to the point where you need testers for reading/writing the syscon chip, I'd love to help.

Pazuzo, 2024-10-11:
Nice, will there be an edition for Pro Micro ?

closed 2024-12-05 by AbkarinoMHM

RustyRaindeer, 2025-02-21:
This has been marked as completed, but I do not see any other option than using the over priced Teensy...

Is there any hope that you release RPI pico version anytime soon, or open the source code for community to work on it? I would love to help too. 

Cheers!!

LouisGarry, 2025-07-24:
> For Arduino Nano we found it unstable because it had only one HW UART interface and to fix it we will need to use more components with the Nano to increase the stability

Hi master, would you mind sharing the ino file? I'd like to try it.