AbkarinoMHM/PS4SysconTools #45: Not connecting to teensy 2.0++

R0PKE123, 2025-04-18:
When launching the programm the tool says it is not connected when it is, and is programmed, but on the older non beta version it is stuck on connecting to programmer, and it shows two COM7 ports?

R0PKE123, 2025-04-22:
fixed neede to use windows 10

Pazuzo, 2025-04-22:
Mine working on 11 just fine :)

R0PKE123, 2025-04-22:
Ok now it connects on a windows 10 virtual machine but it cannot exploit the siscon here is how everything is wired

[image]

[image]

R0PKE123, 2025-04-22:
Ok now it connects on a virtual machine of windows 10 virtualbox, but it cannot exploit the siscon trying to find a way to launch a debugger

[image]

[image]

AbkarinoMHM, 2025-04-22:
Actually my main development PC is Windows 11 based, so i tested everything under Windows 11 and it was working fine.
So please give more details about your issue so i can check.

R0PKE123, 2025-05-04:
it appears that the cable that i am using is not reliable, but i don't know for sure

R0PKE123, 2025-05-04:
if you could what details should I give you?

R0PKE123, 2025-05-07:
The adapter change did not work, it is not suprising considering it is somehow ment for sata but that's besides the point. \Finally got to getting the error on the old versions here it is: See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.IO.IOException: Port not found: COM8
   at PS4_Syscon_Tools.PS4SysconTool.PS4SysconToolConnect(String port, String& version, String& freeMemory, Boolean& debugMode)
   at PS4_Syscon_Tools.frmPS4SysconTools.btnConnect_Click(Object sender, EventArgs e)
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9310.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
PS4 Syscon Tools
    Assembly Version: 0.1.6.0
    Win32 Version: 0.1.6.0
    CodeBase: file:///C:/Users/R0pke/Downloads/PS4SysconTools_v1.60/PC/PS4%20Syscon%20Tools.exe
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9256.0 built by: NET481REL1LAST_B
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9310.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System.Configuration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9297.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
Accessibility
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/Accessibility/v4.0_4.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------
RJCP.SerialPortStream
    Assembly Version: 2.4.1.0
    Win32 Version: 2.4.1.0
    CodeBase: file:///C:/Users/R0pke/Downloads/PS4SysconTools_v1.60/PC/RJCP.SerialPortStream.DLL
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.

R0PKE123, 2025-05-07:
this is another error that is different: See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.UnauthorizedAccessException: Access Denied: COM7
   at PS4_Syscon_Tools.PS4SysconTool.PS4SysconToolConnect(String port, String& version, String& freeMemory, Boolean& debugMode)
   at PS4_Syscon_Tools.frmPS4SysconTools.btnConnect_Click(Object sender, EventArgs e)
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9310.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
PS4 Syscon Tools
    Assembly Version: 0.1.6.0
    Win32 Version: 0.1.6.0
    CodeBase: file:///C:/Users/R0pke/Downloads/PS4SysconTools_v1.60/PC/PS4%20Syscon%20Tools.exe
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9256.0 built by: NET481REL1LAST_B
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9310.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System.Configuration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9297.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
Accessibility
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/Accessibility/v4.0_4.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------
RJCP.SerialPortStream
    Assembly Version: 2.4.1.0
    Win32 Version: 2.4.1.0
    CodeBase: file:///C:/Users/R0pke/Downloads/PS4SysconTools_v1.60/PC/RJCP.SerialPortStream.DLL
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.

R0PKE123, 2025-05-07:
The newer versions just say that it is not connected and when I replug they freeze here: PS4 Syscon Tool Started.
Connecting To PS4 Syscon Tool.

AbkarinoMHM, 2025-05-12:
please give more details about your OS, .NET framework installed, teensy VID/PID on your PC so i can help.

R0PKE123, 2025-05-15:
Well I tested windows 11 the newest version and windows 10, have java, have .net, and I assume I find the VID/PID of my teensy throught device manager, right?

R0PKE123, 2025-05-15:
Ok this is what task manager says: USB\VID_16C0&PID_0483&REV_0272
USB\VID_16C0&PID_0483

AbkarinoMHM, 2025-05-15:
This is the correct VID & PID values for teensy++ 2.0, so it should work, anyway i'm working in a new version right now, i had rewrote teensy detection function with some improvements so just wait and test this new release.

AbkarinoMHM, 2025-05-17:
I beleive that your issue is related to your Windows OS setup, since your main issue now is Access Denid so something blocks access to COM port.
Try this on another PC if you have, or better try to reinstall your Windows OS first, or if you have any Anti Virus application, try to disable it first and run my tool as admin.

R0PKE123, 2025-05-19:
I cannot run the latest release as admin, the other releases still have the same issue and I tested it on my crummy windows 10 laptop that only has one working usb port, and it did not work, will test with the latest release, currently in work

R0PKE123, 2025-05-19:
I might also try to install a generic driver see if that fixes it, becuase taht fixed some stuborn devices

R0PKE123, 2025-05-21:
Ok so trying to run the tool on the windows 10 laptop did not work, and I did not find anything that helped.

R0PKE123, 2025-05-26:
Yes, as I have said I have tested it with two laptops. I am ordering another cable from aliexpress though so maybe that will fix it?

closed 2025-06-06 by R0PKE123