AbkarinoMHM/PS4SysconTools #54: Feature Request: Automatic Backup

water2891, 2025-08-06:
Automatically back up the existing firmware in the chip before writing, to prevent people from forgetting to dump the old firmware like me. I think it can help a lot of careless people.

AbkarinoMHM, 2025-08-06:
Hi,

   Thanks for your suggestion, do you mean auto dump any syscon chip before writing to it?

itsemtiV, 2025-08-06:
> Hi,
> 
> Thanks for your suggestion, do you mean auto dump any syscon chip before writing to it?

exactly, also it would be great if you implement a syscon rebuild future

water2891, 2025-08-07:
> Hi,
> 
> Thanks for your suggestion, do you mean auto dump any syscon chip before writing to it?

Yes, I forgot dump 12.52's syscon and overwrote it with old syscon data. Now the console is bricked ···