AbkarinoMHM/PS4SysconTools #57: Unable to downgrade firmware =>>error: main(3738) checkupdversion

sapabuon, 2025-08-10:
My PS4 slim is on fw 12.50. I downgraded to 12.02. But when I tried all the patches created by ps4 we tools ( A,B,C,D ) my PS4 could not boot. With the error: error: main(3738) checkupdversion 0x12508000 != 0x12020000. Can anyone help me? Thank you very much!
[https://drive.google.com/file/d/1A_vuwIvIwdz0dYwsPYUtLuvII8koweD2/view?usp=sharing](url)

water2891, 2025-08-11:
try patch_B then continue patch_C on your xxx_patch_B.bin, as xxx_patch_B_patch_C.bin

AbkarinoMHM, 2025-08-11:
I will close this issue since it is not related to PS4 Syscon Tools, if you need support for fw patching/reverting, please join our discod server: https://discord.gg/2N9uZFct